bigtreetech/BIGTREETECH-TMC2209-V1.2 #2: BTT SKR1.3+BTT TMC2209 sound & vibration issue

muuus, 2019-11-17:
I have very strange issue with my new SKR 1.3 board and TMC2209 from BBT. Please see attached video with sound ON.

Video: https://youtu.be/lpWGFX8TAkM

StealthChop working in veeery strange way. Terrible vibrations and noise. If I change TMC mode to SpreadCycle, everything became almost normal, but there is still some minor vibrations and flow instability, as result outer walls are not smooth.

Only if I change speed to very low - less then 5 mm\s sound is disappearing.
2209 in UART mode. M122 output below. Latest Marlin 2.0 from git.

Really need some help here. I'm out of ideas T_T Something is really wrong here.

Already tried to change:
MINIMUM_STEPPER_PULSE - no effect
MINIMUM_STEPPER_POST/PRE_DIR_DELAY - no effect
ADAPTIVE_STEP_SMOOTHING - no effect
SQUARE_WAVE_STEPPING - no effect
Latest BTT TMC libraries - no effect
Steps per mm - no effect
Current - 600, 900, 1200 - no effect
LA - no effect

> Send: M122
> Recv: 		X	Y	Z	E
> Recv: Address		0	0	0	0
> Recv: Enabled		true	true	true	false
> Recv: Set current	1200	1200	1200	1200
> Recv: RMS current	1160	1160	1160	1160
> Recv: MAX current	1636	1636	1636	1636
> Recv: Run current	20/31	20/31	20/31	20/31
> Recv: Hold current	10/31	10/31	10/31	10/31
> Recv: CS actual	10/31	10/31	10/31	10/31
> Recv: PWM scale	44	37	41	12
> Recv: vsense		0=.325	0=.325	0=.325	0=.325
> Recv: stealthChop	true	true	true	false
> Recv: msteps		16	16	16	16
> Recv: tstep		max	max	max	max
> Recv: pwm
> Recv: threshold	197	197	197	128
> Recv: [mm/s]		50	50	50	30
> Recv: OT prewarn	false	false	false	false
> Recv: OT prewarn has
> Recv: been triggered	false	false	false	false
> Recv: off time	4	4	4	4
> Recv: blank time	24	24	24	24
> Recv: hysteresis
> Recv: -end		2	2	2	2
> Recv: -start		1	1	1	1
> Recv: Stallguard thrs	0	0	0	0
> Recv: DRVSTATUS	X	Y	Z	E
> Recv: stst		*	*	*	*
> Recv: olb
> Recv: ola
> Recv: s2gb
> Recv: s2ga
> Recv: otpw
> Recv: ot
> Recv: 157C
> Recv: 150C
> Recv: 143C
> Recv: 120C
> Recv: s2vsa
> Recv: s2vsb
> Recv: Driver registers:
> Recv: 		X	0xC0:0A:00:00
> Recv: 		Y	0xC0:0A:00:00
> Recv: 		Z	0xC0:0A:00:00
> Recv: 		E	0x80:0A:00:00
> Recv: 
> Recv: 
> Recv: Testing X connection... OK
> Recv: Testing Y connection... OK
> Recv: Testing Z connection... OK
> Recv: Testing E connection... OK
> Recv: ok

muuus, 2019-11-20:
Solved it. Moved to another TMC repo in platformio.ini:
```
[env:LPC1768]
platform          = https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip
framework         = arduino
board             = nxp_lpc1768
build_flags       = -DU8G_HAL_LINKS -IMarlin/src/HAL/HAL_LPC1768/include -IMarlin/src/HAL/HAL_LPC1768/u8g ${common.build_flags}
# debug options for backtrace
#  -funwind-tables
#  -mpoke-function-name
lib_ldf_mode      = off
lib_compat_mode   = strict
extra_scripts     = Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_LPC1768>
monitor_speed     = 250000
lib_deps          = Servo
  LiquidCrystal
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  #TMCStepper=https://github.com/p3p/TMCStepper/archive/pr_lpctimingfix.zip #not ok
  TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip #ok
  Adafruit NeoPixel=https://github.com/p3p/Adafruit_NeoPixel/archive/release.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
```

closed 2019-11-20 by muuus

freighttt, 2020-06-14:
I have this same problem, the repository didn’t solve it though.

gretzke, 2020-11-04:
did you manage to solve it?

muuus, 2020-11-04:
> did you manage to solve it?

yeah :) 
The main issue was with ferrite filters. I use them as additional filter between the mainboard and step motors. After I removed them and moved to another repo, everything started work without problems.