bigtreetech/BIGTREETECH-TMC2209-V1.2 #3: BTT v1.2 TMC2209 Sensorless Homing not working

buddybu, 2019-11-20:
I have an SKR v1.3 and I purchased the BTT v1.2 TMC2209.  I did not solder any connections (not sure I have to), and I can get each of the motors moving by using the controls, but when I want to do the homing, it is not working.  I have attached my configuration files here

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/files/3867142/Marlin.zip)

buddybu, 2019-11-24:
Anyone have any information that could help me?

cameronsocialhardware, 2020-04-10:
@buddybu I've managed to get it working for X and Y on the SKR PRO V1.1 but Z is very difficult with lead screws because of the torque. I've uploaded my config etc. here so hopefully that will help you: https://github.com/cameronsocialhardware/Marlin-2.0-Sensorless-Homing

buddybu, 2020-04-10:
Thank you @cameronsocialhardware.  I actually moved to the SKR v1.4, and have everything working.  Mine is a refurbished 3DSystems Cube Generation 2 whose motherboard had died.  I have since replaced the hot end and adapted a E3D v6 clone to the printer as well.  This printer has a fixed x axis at the top of the printer.  The Y-Axis is attached to the Z-Axis and is raised to meet the head at the top of the printer.  It was difficult to get it going, but I managed to accomplish it.  I have to flip every one of my models, as by inverting the Z axis, everything prints backwards.  :)

closed 2020-04-10 by buddybu