bigtreetech/BIGTREETECH-TMC2209-V1.2 #4: Microstep charts in manual and schematic do not agree

Jumbu, 2020-02-25:
https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/manual/TMC2209-V1.2-manual.pdf
https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf
The microstep MS1 and MS2 values do not line up between these. I'd like to be able to set up whole stepping or half stepping but this may or not be possible depending on which chart is correct.

maxd, 2020-02-29:
Looks like `TMC2209-V1.2-manual.pdf` contains right settings. 

Here is MS1/MS2 settings from [TMC2208 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf):

[image]

and here is from [TMC2209 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf):

[image]

As you can see they have different settings for MS1/MS2 and settings from TMC2209 datasheet correspond to settings from `TMC2209-V1.2-manual.pdf` manual.

@bigtreetech Am I right? Could you please fix table with microstep settings in `Schematic/TMC2209-V1.2.pdf`?

Jumbu, 2020-02-29:
Oddly enough, I used the table in the schematic to get half stepping and it works. It must be correct.

-------- Original Message --------

nmaas87, 2023-09-04:
Tested this today again with the TMC2209 version 1.3 stepper drivers. The TMC2209 datasheet is correct in that fact, the TMC2209 does not have the option to be set to 1/2 or 1/4 step, the base (floating, internally pulled to ground) setting is 1/8 and the most coarse setting that can be achieved via MS1/MS2 settings. The [TMC2209-V1.2.pdf](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf) is with that wrong, at least for the v.1.3 boards (and I guess for the v.1.2 also if you don't OTP the drivers...)