bigtreetech/BIGTREETECH-TMC2209-V1.2 #7: SKR 1.4 Turbo TMC2209 Error: All LOW

dezodude, 2020-06-07:
M112
Recv: Driver registers:
Recv: 		X	0x00:00:00:00	 Bad response!
Recv: 		Y	0xC0:0B:00:00
Recv: 		Z	0xC0:0B:00:00
Recv: 		E	0xC0:0C:00:00
Recv: 
Recv: 
Recv: Testing X connection... Error: All LOW
Recv: Testing Y connection... OK
Recv: Testing Z connection... OK
Recv: Testing E connection... OK

I have tried new stepper motor cable, same result
I have wapped X and Y driver, same result

Using Latest Marlin bugfix build as of 07/06/2020 - Looking for support

rals8n, 2020-06-27:
yeah this was the bane of my existance too. what you need to do is run the M122 S1 {S1 focuses on the steppers in particular} and remember which axis is reading bad. Then take the stepper driver and swap it with another axis, doesnt which. Run the M122 S1 again, if the stepper reads error on the axis you moved it to. The stepper driver is defective and there is no fix. Its junk. cut your losses and buy another one of the same type to replace it. no amount time or bugfix will do anything. bigtreetech is a complete joke as far as quality control is concerned. you can try and get a refund for it but just order the stepper 1st cuz it will take forever for them to get back to you. youre looking at atleast a month if youre lucky. So youre better off to just get it now so you can get it running. In fact getting a couple would be a good idea cuz the odds of getting another crapped out part are very high. Then youre not sitting on your thumbs waiting a month plusfor them to get back to you and anotgher couple weeks for shipping. This company is the eppidemy of made in china.... I will never get anything from these guys EVER AGAIN. they make it sound amazing but their stuff is anything but. so yeah its nothing to do with your setup. its the hardware. i wish i had better news for you. I went through the same exact problem and it was a complete nightmare to say the least. dont waste anymore time. there is no fix.

dezodude, 2020-06-27:
Workaround implemented and post updated

rals8n, 2020-06-30:
yeah its insanely frustrating...I got an skr1.4 and spent a week trying to set it up and nothing worked. I got it about a week after their release. So there was nothing out there to help. So I was going off the joke of a manual. I eventually tried contacting them. None of the phone numbers worked and no response to the email for almost a month. Once I did get a message it said it was defective if no green light came on when heating. Which it wasn't a week later after saying it was defective they sent me the turbo version and told me to just use the the 2209s that came with the defective one. Which 2 ended up being messed up because of the defective board. So had to order those. And instead of returning those too, I just bought them because I didn't want to wait 3 more months by dealing with the refund. I ended up getting the replacement board and got all the firmware setup and could not get any motion to the axis. spent about a week trying to figure out why. Now there was finally some people out there who got it all going and used there firmware bin file who had the exact same setup. To find it wasnt the 2209s but the new replacement board was defective too. I got another one from them and this time I just dropped it in and it worked perfect....it was such a pain in the ass and no joke took a little over 2 months. But with no real references it made it a nightmare cuz I didn't know if I was doing something wrong. And I'm pretty sure support doesn't actually exist....these things have so much to offer...but wtf. And yeah man there is a 2 week curse with these things, so you having issues every couple weeks seems to be really common. Luckily yours works after cuz most just crap out. It would be one thing if these were rare issues but there are so many people with issues. How one board drivers have thousands of bufixes pretty much says it all lol

Eddiiie, 2020-08-18:
Wow.  I have used Marlin 1.x in all my printers for the last 5 years.  Decided to try the TMC 2208 on my RAMPS and Arduino... Pulled my hair out, decided to go to 32 bit and this SKR 1.4 Turbo with the TMS2209's, 

I am in the process of moving all the wires over to the new board, moved to Marlin 2 bug fix, jumped some hoops getting it to compile.   Plugging in the USB cable from my computer the LCD says TMC CONNECTION ERROR...  Is this the same problem you have?  OR is this something different?    It doesn't tell me which driver has the connection error, probably all of them.  

I am confused if I need to power the board to avoid this message.  If so, great.  I am just not there yet.    Took my all day/night to get Marlin to even compile and make the BIN file.    Was going to open a new issue but saw this and thought I'd ask.

Thank you!

Eddiiie, 2020-08-18:
Also, do I need to move the UART pin from Pin 4 (factory default) to pin 5?    Is there a jumper wire needed?  No, right?  It is part of the beauty of the SKR1.4 Turbo.. Right?

Eventually, when I connect power, I need to adjust the potentiometer on the driver, right?

JoeEnderman, 2021-01-11:
I had the same issue, except swapping the drivers left the problem where it was. I bought the board because it offered five drivers versus the e3 v2's four. I wanted a second extruder, so choices were limited. I know the drivers are fine because x and e1 read as ok no matter what driver was where when the m122 or m122 s1 command was run. I am trying to get a new mainboard because I am literally going to fry an extruder motor if I don't.

rals8n, 2021-01-11:
I had returned all 5 drivers and had the new ones installed with the 1
jumper for sensorless homing a viola all was good and worked great. No clue
how I ended up with 5 brand new driver that happened to be junk but that
was indeed the case. It left me with a love hate relationship for btt cus
there was nothing done on there behalf to try and help. Just alot of
shipping and waiting time from Amazon lol

Heratiki, 2021-02-05:
I'm now having this issue where X is Ok but Y Z and E0 all show Error All LOW. I've tried initializing the steppers with M17. I've tried moving them but they all still read the same thing.  I have no clue what is going on and after ordering 5 more step sticks and the issue still being present I'm starting to hate my damn printer.  Why is it this hard to get anything working?

Prophet714, 2021-12-23:
@Heratiki I'm running into the same issue you did here. Were you able to solve it?

pman92, 2023-01-03:
I had exactly the same problem. Same board, same drivers, same X axis not working problem. Confirmed it as a hardware problem by changing pins file the same. 

I found that the first pin along on the driver socket hadn't been soldered on underneath. It was completely disconnected. Rubbish quality control. Luckily soldering it on fixed the problem for me. 

https://forum.v1engineering.com/t/btt-quality-control/35959