bigtreetech/BIGTREETECH-TMC2209-V1.2 #17: TMC2209 V1.3 entering spreadcycle mode

ussforbes, 2023-05-17:
Hello, I will be using a teensy 4.1 and 2 TMC2209 V1.3s wired directly without a shield.  

For the 2209 V1.3, is spreadcycle the default mode?  If not, how do I switch it?  

If I just have to use another pin on the teensy to control it or if I have to do the OTP. I can't find any info on this as some sources say it's in spreadcycle by default and all otp instructions are for 2208 or v1.2 which have different solder pad configurations than the 1.3... 

Mixed conflicting information is what I'm finding so I figured I'd go straight to the source.

srizvan, 2023-09-14:
I'm also interested in the same information. I see from the manual for V1.2 that it's in stealthchop mode by default (https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/manual/TMC2209-V1.2-manual.pdf page 8). 
If you want to enable spreadcycle, move the resistor on SPRE contacts. On V1.3 there are no such contacts...

If anyone has information on how to enable it on 1.3, please share the information.

lord-BUSV, 2024-09-26:
Hi there!
I've just bought a couple of TMC2209 V1.3 and I discovered that these contacts are connected to SPREAD pin on main chip (pin #7) and to Vcc (VIO pin). Moreover, at (https://www.programming-electronics-diy.xyz/2023/12/tmc2209-stepper-driver-module-tutorial.html#Setting_up_CoolStep) is specified that the SPREAD pin has a internal resistor to pull the pin down (for this reason it is possible to leave the pin unconnected when the chopper-mode is selected). I measured this R and it seems to be 150KOhm, so I'm going to put a small resistor (about 1kOhm - 4.7kOhm) between these contacts to pull-up the TMC2209's pin and to enable the spread-cycle mode.
[image]

jkexbx, 2025-07-15:
Do you know of anyway to do this with programming? Underestimated how small that resistor is