bigtreetech/BIGTREETECH-TMC2209-V1.2 #18: Documentation for V1.3 boards

elbobius, 2024-03-09:
I could not find any manual/hardware documentation for the V1.3 driver boards. The boards layout differs from V1.2 and the soldering jumper used to enable UART/PDM seems to be gone with V1.3. Do the stock boards come with enabled uart? Adding some info for V1.3 would be great.

hofingermarkus, 2025-02-16:
I think I found it here:

https://github.com/bigtreetech/BIGTREETECH-Stepper-Motor-Driver/tree/master/TMC2209

Garywoo, 2026-02-19:
The manual does not label the VREF and INDEX pins, which is what I'm trying to find out. Does anyone know?

Pin 1: _???_
Pin 2: DIAG
Pin: 3: _???_

[image]